nkinnan/esphome-pace-bms #1: schematic for conection?

rflx18, 2024-08-30:
can you please share the schematic for connecting? i have tried with rs485 without sucess :(

nkinnan, 2024-08-30:
My component isn't complete yet, and I haven't pushed to main yet.  I assume you're using `external_component`?  If you're pointing to the initial_commit branch it will break on and off as I am in active development.  Happy to let you know when it's ready, probably only a couple more days.

If you're using a different component or rolling your own, and just want to know how to connect to RS485:
1) declare in YAML the UART component for your ESP including both TX and RX pins
2) physically connect UART TX pin to serial/rs485 converter chip's RX pin
3) physically connect UART RX pin to serial/rs485 converter chip's TX pin
4) declare in YAML your ESP component that speaks the rs485 protocol (such as this one for paceic) and set set it's flow control pin (rs485 uses two wires shared for transmit/receive so a flow control output is required)
5) physically connect the flow control pin to the chip's RX_Enable and TX_NotEnable pins
6) physically connect the rs485 converter chip's A/B (D+/D-) pins to your RS485 bus.  For a Pace BMS that's the two right-most pins on the ethernet connector (not socket) if the clip is facing away from you and the wire is hanging down.

I'll have a full schematic, pictures, and instructions in easier to read format once this is ready for consumption, probably a week or so.

nkinnan, 2024-08-30:
You may find the diagrams here helpful, they are for a component speaking a slightly different version of paceic: https://github.com/syssi/esphome-seplos-bms

nkinnan, 2024-09-20:
I've updated the readme.md with full instructions and diagrams on wiring up either a RS485 or RS232 adapter between an ESP and the PACE BMS.

closed 2024-09-20 by nkinnan