nkinnan/esphome-pace-bms #24: EASUN iBattery-EA-51.2V-200AH - two batteries connected in parallel

f3nix, 2024-10-08:
Hi.
I have two batteries connected in parallel with an RS485 link.
First one is set to address 1. Second one is set to address 2.
I have connected one esp32 to each one.

The one with address 1 works as expected.

The second one has a problem and reads only zeroes.

```
>>> 7E:32:35:30:32:34:36:34:32:45:30:30:32:30:32:46:44:32:45:0D
<<< 7E:32:35:30:32:34:36:30:30:31:30:39:36:30:30:30:32:31:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:36:30:41:41:41:30:41:41:41:30:41:41:41:30:41:41:41:30:41:41:41:30:41:41:41:30:30:30:30:30:30:30:30:30:30:30:30:30:39:30:30:30:30:30:30:30:30:30:30:30:30:36:34:30:30:30:30:30:30:30:30:30:30:30:30:30
```

I have peeked how pbms tools does it.

Connected to RS232 - Battery 1 - reading analog info of battery 1
```
7e 32 35 30 31 34 36 34 32 45 30 30 32 30 31 46 44 33 30 0d
```

Connected to RS232 - Battery 1 - reading analog info of battery 2
```
7e 32 35 30 31 34 36 34 32 45 30 30 32 30 32 46 44 32 46 0d
```

Connected to RS232 - Battery 2 - reading analog info of battery 1 (battery 2 really) (15th byte is different compared to what we are doing now)
```
7e 32 35 30 32 34 36 34 32 45 30 30 32 30 **31** 46 44 32 46 0d
```

Answer for the third case is:
```
7e 32 35 30 32 34 36 30 30 31 30 39 36 30 30 30
31 31 30 30 43 46 46 30 43 46 45 30 43 46 45 30
43 46 45 30 43 46 46 30 43 46 45 30 43 46 46 30
43 46 46 30 43 46 46 30 43 46 46 30 43 46 46 30
43 46 46 30 43 46 46 30 43 46 45 30 43 46 46 30
43 46 45 30 36 30 42 34 39 30 42 34 37 30 42 34
36 30 42 34 38 30 42 35 44 30 42 35 34 46 45 45
41 43 46 45 41 34 36 32 32 30 39 35 30 46 35 30
30 30 33 34 45 32 30 35 37 34 45 32 30 34 45 32
30 34 45 32 30 34 45 32 30 36 34 44 31 34 42 30
30 30 30 44 42 32 44 0d
```

Cheers,
Mateusz

nkinnan, 2024-10-08:
The component as-written does not currently support chained batteries.  If you have them chained so that your inverter can talk to both at the same time, this component might work if you connect it to the RS232 port on each instead, but the RS485 port will not work properly when they are also connected over RS485.  Plus the inverter will be sending RS485 commands the same time the ESP is sending RS485 commands causing collisions due to conflicting "bus masters".

nkinnan, 2024-10-08:
I don't have access to chained batteries to even test that functionality right now if I did want to support it unfortunately.  Probably best to not connect them.  Unless you need to for your inverter, in which case I'm not sure what the best solution would be.

nkinnan, 2024-10-08:
If you want to try the RS232 port on each (while they are chained) try sending each it's own address (to bypass the chaining functionality) there's a chance it'll work properly in that case.  Let me know what you find.

f3nix, 2024-10-08:
> If you want to try the RS232 port on each (while they are chained) try sending each it's own address (to bypass the chaining functionality) there's a chance it'll work properly in that case. Let me know what you find.

Here is my current setup. I want separate esp's to be able to change settings in each bms. As far as I know it is not possible to do with one connected to the first battery in the chain.

[image]

The issue is about this case.

Config of second esp32:
```
pace_bms:
  id: pace_bms_at_address_2
  address: 2
  uart_id: uart_0
  # needed for RS485
  # flow_control_pin: GPIO5 
  update_interval: 20s
  request_throttle: 50ms 
  response_timeout: 2000ms
```

Changing the 15th byte to 31 should work in this case. Will have to test it tonight.

Cheers,
Mateusz

nkinnan, 2024-10-09:
Where are you getting those hex bytes from? It would be helpful if you turn on VERY_VERBOSE log level and show the string sent to/from the BMS as displayed by the component.

f3nix, 2024-10-09:
In the first post the first code frame contains uart debug data as seen in the console (which does not work).

The following three commands in first post were captured with ioninja. pbmstools <-> ioninja <-> bms.

Anyway. I have changed the code a little bit to send another request to bms number 2 and it basically works :) (quick and dirty hack). So now it seems that I can read data from both of them with separate esp's.

```

[05:58:49][D][pace_bms:433]: Sending 'read analog information' request
[05:58:49][D][uart_debug:114]: >>> 7E:32:35:30:32:34:36:34:32:45:30:30:32:30:31:46:44:32:46:0D
[05:58:49][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:30:30:31:30:39:36:30:30:30:31:31:30:30:43:44:36:30:43:44:35:30:43:44:34:30:43:44:35:30:43:44:35:30:43:44:35:30:43:44:36:30:43:44:35:30:43:44:34:30:43:44:35:30:43:44:36:30:43:44:34:30:43:44:36:30:43:44:34:30:43:44:35:30:43:44:35:30:36:30:42:36:35:30:42:36:34:30:42:36:33:30:42:36:34:30:42:37:42:30:42:37:31:46:45:44:30:43:44:35:30:33:33:46:41:30:39:35:30:46:35:30:30:30:33:34:45:32:30:34:30:34:45:32:30:34:45:32:30:34:45:32:30:34
[05:58:49][D][pace_bms:480]: Processing 'read analog information' response
[05:58:49][E][pace_bms_protocol:022]: Response from wrong bus Id in payload - IGNORED
[05:58:49][W][pace_bms_protocol:025]: Response contains a constant with an unexpected value, this may be an incorrect protocol variant
[05:58:49][E][pace_bms_protocol:022]: Length mismatch reading analog information response: 28 bytes off - IGNORED

[05:58:49][D][pace_bms:433]: Sending 'read status information' request
[05:58:49][D][uart_debug:114]: >>> 7E:32:35:30:32:34:36:34:34:45:30:30:32:30:31:46:44:32:44:0D
[05:58:49][D][pace_bms:496]: Processing 'read status information' response
[05:58:49][E][pace_bms_protocol:022]: Response from wrong bus Id in payload - IGNORED
[05:58:49][E][pace_bms_protocol:022]: Length mismatch reading status information response: 2 bytes off - IGNORED
[05:58:50][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:30:30:45:30:34:45:30:30:30:31:31:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:36:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:45:30:30:30:30:30:30:30:30:30:30:30:30:30:30:45:45:43:32:0
```

Could you please change the address of your bms to 2 and check if it behaves the same?

Cheers,
Mateusz

P.S.
I just came across this info. These are the protocol versions PbmsTools 3.13 speaks with:
```
  public enum _232ProtocalTypeEnum
  {
    [Description("HS-PACE-232-BP_V2.2")] HS_PACE_232_BP_V22,
    [Description("BS-PACE-232-BK_V1.0")] BS_PACE_232_BK_V1,
    [Description("HS-PACE-232-BP_V2.9")] HS_PACE_232_BP_V29,
    [Description("HS-PACE-232-BP_V2.8")] HS_PACE_232_BP_V28,
    [Description("HS-PACE-232-BP_V2.5")] HS_PACE_232_BP_V25,
    [Description("BS-PACE-232-JBS_V1.0")] BS_PACE_232_JBS_V1,
    [Description("PACE-RS232-TY16S")] PACE_RS232_TY16S,
    [Description("HS-PACE-232-BP_V2.7")] HS_PACE_232_BP_V27,
    [Description("HS-PACE-232-BP_V2.6")] HS_PACE_232_BP_V26,
    [Description("HS-PACE-232-RD_V1.0")] HS_PACE_232_RD_V1,
    [Description("HS-PACE-232-BP_V2.4")] HS_PACE_232_BP_V24,
    [Description("HS-PACE-232-BP_V2.1")] HS_PACE_232_BP_V21,
    [Description("HS-PACE-232-BP_V1.6")] HS_PACE_232_BP_V16,
    [Description("HS-PACE-232-BP_V1.6F")] HS_PACE_232_BP_V16FFFF,
    [Description("HS-PACE-232-LISHI_V1.0")] HS_PACE_232_LISHI_V1,
    [Description("HS-PACE-232-LISHI_V1.1")] HS_PACE_232_LISHI_V11,
    [Description("HS-PACE-232-BP-KX_V1.0")] HS_PACE_232_BP_KX_V1,
    [Description("HS-PACE-232-BP-KX_V1.0F")] HS_PACE_232_BP_KX_V1FFFF,
    [Description("HS-PACE-232-BP-PND_V1.0")] HS_PACE_232_BP_PND_V1,
    [Description("HS-PC-232-BP-YY_V1.1")] HS_PC_232_BP_YY_V11,
    [Description("HS-PACE-232-DC_V1.0")] HS_PACE_232_DC_V1,
    [Description("HS-PACE-232-XY_V1.1")] HS_PACE_232_XY_V11,
    [Description("HS-PACE-232-XY_V1.0")] HS_PACE_232_XY_V1,
    [Description("HS-PACE-232-ZT_V1.0")] HS_PACE_232_ZT_V1,
    [Description("HS-PACE-232-OLT_V1.0")] HS_PACE_232_OLT_V1,
  }
```

f3nix, 2024-10-10:
#25
@nkinnan wrote 

> We should chat separately about your other modification. What you're doing by hard-coding 01 in the body of the request is saying "from the BMS at address 2, request the data from address 1". If you double-check your readings, you may see you're getting duplicate values (all values are from address 1).

I do not believe this is true. Here is a screen from the protocol description:

[image]

When connecting like this:
```
  ESP32
    |
 -------
| BMS 1 |-\ PACK 1
 -------  |
          |
 -------  |
| BMS 2 |-/ PACK 2
 -------
```

I get this:
```
7e 32 35 30 31 34 36 39 30 30 30 30 30 46 44 41  ~250146900000FDA
35 0d                                            5♪
7e 32 35 30 31 34 36 30 30 45 30 30 32 30 32 46  ~25014600E00202F
44 33 35 0d                                      D35♪
```

And when connecting like this:
```
 -------
| BMS 1 |-\ 
 -------  |
          |
 -------  |
| BMS 2 |-/ PACK 1
 -------
    |
  ESP32
```

I get this:
```
7e 32 35 30 32 34 36 39 30 30 30 30 30 46 44 41  ~250246900000FDA
34 0d                                            4♪
7e 32 35 30 32 34 36 30 30 45 30 30 32 30 31 46  ~25024600E00201F
44 33 35 0d                                      D35♪
```

I have double checked the values on the lcd screens of both batteries and esphome gets good values from both of them.

So addr != pack number.

So the code should be changed I think.
For example:
```
bool PaceBmsProtocolV25::CreateReadAnalogInformationRequest(const uint8_t busId, std::vector<uint8_t>& request)
{
	// the payload is the requested busId (could be FF for "get all" when speaking to a set of daisy-chained units but this code doesn't support that)
	const uint16_t payloadLen = 2;
	std::vector<uint8_t> payload(payloadLen);
	uint16_t payloadOffset = 0;
```
this busId should be packId
```
	WriteHexEncodedByte(payload, payloadOffset, busId);
```

   and this busId is the real bms addr.
```
	CreateRequest(busId, CID2_ReadAnalogInformation, payload, request);
```

```
	return true;
}
```

Cheers,
Mateusz

P.S. You could probably test it by setting the address of you bms to a value bigger than 1. It works here even if I disconnect the communication cable between the two bms'

f3nix, 2024-10-10:
Here's a screen with some values of the sensors:

[image]

Cheers,
Mateusz

nkinnan, 2024-10-11:
Interesting, I'm interpreting that protocol doc a bit different, but I can't argue with your results.  I'll run some tests with my BMS and see if I can figure out or confirm what's happening after I push a quick update to handle the "user defined" UD value 9 so it handles your protocol variation more gracefully.  After that we can hopefully get you squared away with your multi-pack situation.  Your debugging this is quite helpful, thank you :)

MichaelEFlip, 2025-03-16:
We have here three FSP PS5120E connected in parallel via RS485 link port (see https://github.com/nkinnan/esphome-pace-bms/issues/31 for basic working information).
Inverter is connected via RS485 master port.
ESP32 is connected via RS232 console port.

From PBMSTools I am able to query the information from the battery pack 2 and 3 while being connected via RS232 to battery pack 1.
I am also interested in reading sensor values from parallel connected batteries using only one ESP32.

I can support in capturing RS232 logs from PBMSTools or own queries.
Also I could support in trying to change the relevant code.

What do you need?

RoganDawes, 2025-03-16:
I too have multiple (2) packs, connected to each other via RS485. I am currently using tertiush/pacebms addon via the RS232 port, which can read both packs without issue, but would like to switch to this ESPHome component for access to the RW configuration registers.

> This component currently requires one UART per BMS, though I'm considering a design change that would allow it to read "daisy chained" BMSes in the future.

I'm also interested in solving this problem. Do you have any details of the design change that would be needed? Happy to test/debug as needed. Is it as simple as changing the 15th byte to indicate the offset of the pack from the one currently connected to, and being addressed in byte 5?

> 7e 32 35 30 31 34 36 34 32 45 30 30 32 30 32 46 44 32 46 0d

In which case, the important configuration parameters would be "id of battery that the ESP32 is physically connected to", and "number of hops to jump through"+1 (i.e. 0 hops to talk to the directly connected battery+1 to get 0x31 in byte 15, 1 hop to talk to the next battery +1 => 0x32, etc). Would be interesting to see whether that applies if e.g. there is a pair of batteries numbered 1 and 3, i.e. skipping one of the possible packs.

SaschaKP, 2025-09-18:
I've currently modified the esphome-seplos-bms of syssi and adapted it to work with pace with rs232 for daisy chained batteries, by reading and making a small adaptations of it to make it work with pace bms and multiple packs, of course you can't change parameters, since the code of syssi is pretty minimal.

references:
original: https://github.com/syssi/esphome-seplos-bms
modified for pace: https://github.com/SaschaKP/esphome-pace-bms