nkinnan/esphome-pace-bms #25: Tewaycell 20kWh all-in-one support 0x25

tbrasser, 2024-10-10:
Hey, awesome component. I think I have the same problem as the other issue, how would I ignore those extra bytes?

```
[13:49:08][VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=14665392 (now=14670392) [13:49:08][V][pace_bms:105]: Queueing v25 refresh commands [13:49:08][V][pace_bms:295]: Read commands queued: 21 [13:49:08][D][pace_bms:433]: Sending 'read analog information' request [13:49:09][V][pace_bms:456]: Processing response frame for 'read analog information' request [13:49:09][D][pace_bms:480]: Processing 'read analog information' response [13:49:09][W][pace_bms_protocol:025]: Response contains a constant with an unexpected value, this may be an incorrect protocol variant [13:49:09][E][pace_bms_protocol:022]: Length mismatch reading analog information response: 28 bytes off [13:49:09][E][pace_bms:485]: Unable to decode 'read analog information' response [13:49:09][D][pace_bms:433]: Sending 'read status information' request [13:49:09][V][pace_bms:456]: Processing response frame for 'read status information' request [13:49:09][D][pace_bms:496]: Processing 'read status information' response [13:49:09][E][pace_bms_protocol:022]: Length mismatch reading status information response: 2 bytes off [13:49:09][E][pace_bms:501]: Unable to decode 'read status information' response [13:49:09][D][pace_bms:433]: Sending 'read hardware version' request [13:49:09][V][pace_bms:456]: Processing response frame for 'read hardware version' request [13:49:09][D][pace_bms:512]: Processing 'read hardware version' response [13:49:09][V][text_sensor:013]: 'Hardware Version': Received new state P16S200A-21587-2.03 [13:49:09][D][text_sensor:064]: 'Hardware Version': Sending state 'P16S200A-21587-2.03' [13:49:09][VV][api.service:174]: send_text_sensor_state_response: TextSensorStateResponse { key: 370526340 state: 'P16S200A-21587-2.03' missing_state: NO } [13:49:09][D][pace_bms:433]: Sending 'read serial number' request [13:49:09][V][pace_bms:456]: Processing response frame for 'read serial number' request 

```

tbrasser, 2024-10-10:
I saw @f3nix his fork and tried that, it works!

f3nix, 2024-10-10:
> I saw @f3nix his fork and tried that, it works!

@tbrasser 
This is just a quick and dirty proof of concept. I would not recommend using it for anything else than testing.

Could you please add this to the "uart: section" and post the logs?

```
    debug:
      direction: BOTH
      dummy_receiver: false
```

Cheers,
Mateusz

nkinnan, 2024-10-10:
@tbrasser @f3nix I will have an update later today that accounts for the length mismatch and allows this component to work with your BMSes but the other change to hard-code address 1 into the body of the requests done by f3nix may work for his situation but is not a good general solution.

tbrasser, 2024-10-10:
> @tbrasser
> This is just a quick and dirty proof of concept. I would not recommend using it for anything else than testing.
>
> Could you please add this to the "uart: section" and post the logs? [...]

here you go :) https://pastebin.com/raw/h9uD2rSg

nkinnan, 2024-10-10:
Setting debug logs to VERY_VERBOSE will also print the strings sent to/from the BMS, but in native ASCII format rather than hex-coded-hex (two levels of indirection).  I would be interested in VERY_VERBOSE logs from you both actually :)

tbrasser, 2024-10-10:
The pastebin link above is with level VERY_VERBOSE

nkinnan, 2024-10-10:
> The pastebin link above is with level VERY_VERBOSE

I don't believe it is :)  If it was, it would be activating this line (and one other) which prints the raw request/response: https://github.com/nkinnan/esphome-pace-bms/blob/f2402f8ddbeb515a70c37be4a1a1b91cb49e5d6c/components/pace_bms/pace_bms_component.cpp#L434

You want this in your YAML:
```yaml
logger:
  level: VERY_VERBOSE
  ```

f3nix, 2024-10-10:
@tbrasser 's bms has the "user define number P" set to 9 too.
```
30:39:
41:30:37:34: 41076 (full capacity)
30:30:30:34: 4     (cycle times)
39:38:35:38: 39000 (design capacity)
30:46:39:38: 3992  (???)
35:38:39:38: 22680 (???)
35:38:39:38: 22680 (???)
35:38:39:38: 22680 (???)
35:38:36:34: 22628 (???)
43:39:43:38: 51656 (independent voltage)
30:30:30:30: 0     (independent current)
```

nkinnan, 2024-10-10:
OK, I'll have an update later today.  We should chat separately about your other modification.  What you're doing by hard-coding 01 in the body of the request is saying "from the BMS at address 2, request the data from address 1".  If you double-check your readings, you may see you're getting duplicate values (all values are from address 1).

tbrasser, 2024-10-10:
> I don't believe it is :)  If it was, it would be activating this line (and one other) which prints the raw request/response: [...]

Weird, see lines 25/26 from my logs:
[20:20:13][C][logger:185]: Logger:
[20:20:13][C][logger:186]: Level: VERY_VERBOSE

nkinnan, 2024-10-11:
> Weird, see lines 25/26 from my logs: [20:20:13][C][logger:185]: Logger: [20:20:13][C][logger:186]: Level: VERY_VERBOSE

That's strange.  How are you reading the logs?  If it's from the embedded web_server dashboard, that display can drop logs at high load.  It'd be better to read the logs directly over serial, or failing that, to limit the number of sensors publishing to only a handful since VERY_VERBOSE spits out a *lot* of logs and could easily overwhelm the web_server.  You should see something like this:

```[21:47:59][D][pace_bms:433]: Sending 'read analog information' request
[21:47:59][VV][pace_bms:437]: Request frame: ~25014642E00201FD30
[21:47:59][VV][uart.arduino_esp32:194]:     Flushing...
[21:47:59][V][pace_bms:456]: Processing response frame for 'read analog information' request
[21:48:00][VV][pace_bms:460]: Response frame: ~25014600F07A0001100CF70CF70CF80CF70CF80CF70CF70CF80CF70CF70CF80CF70CF70CF70CF70CF7060B9E0B9F0BA00B9F0BA20BABFF60CFD825AB03286A00A02710E1A5```

Anyway, it probably doesn't matter too much at this point, I would mostly just like to see the read analog information and the read status information request/response frames in this format (rather than raw hex bytes) for documentation purposes in the code.

----------

I've just updated the code to (I believe) take into account the minor variation in protocol used by your ( @tbrasser ) and @f3nix 's BMSes.  https://github.com/nkinnan/esphome-pace-bms/pull/26  

If you point back to this github for the pace_bms component, would you let me know if all the warnings have gone away and it works as expected?

I'll follow up with f3nix on his dual BMS issue separately.  Let me know if either of you hit any issues otherwise.  Thanks!

Also, I'd appreciate if you could provide any more info on your Tewaycell wall-mount unit.  I'd like to add it to the supported BMS list.  The link to the manufacturers page helps a lot, but if there's any kind of model number or manufacturer's sticker on the unit anywhere, would you mind taking a picture of it for me?

f3nix, 2024-10-11:
P16S200A-21587-2.03

Cheers,
Mateusz

tbrasser, 2024-10-11:
P16S200A-21587-2.03

[image]

I was viewing logs via the esphome dashboard, will try serial (over usb) logging soon.

Working great so far! Thanks!

closed 2024-10-12 by tbrasser