nkinnan/esphome-pace-bms #29: ANENJI 51.2V 100Ah Powerwall LiFePO4 Wall-Mounted Battery Pack 48V 5Kwh

Marcin685, 2025-02-09:
Witam,
Próbuje podłączyć BMS z tej baterii do ESP8266, ale próba niestety nie wychodzi. Robię wszystko według instrukcji, ale nie udaje się żadnych danych odczytać. Podsłuchałem jak komunikuje się oryginale oprogramowanie i załączam.

` 7E 31 32 30 31 34 36 30 30 36 30 32 38 35 32 35   ~120146006028525
 33 33 32 33 33 33 32 32 44 33 30 33 30 33 30 33   33233322D3030303
 31 32 30 32 30 32 30 30 30 30 30 30 30 30 30 30   1202020000000000
 30 30 30 30 30 46 35 44 41 0D 7E 31 32 30 31 34   00000F5DA.~12014
 36 30 30 31 30 33 43 33 30 33 38 33 38 33 32 32   600103C303838322
 44 33 30 33 30 33 33 37 36 33 31 32 45 33 30 33   D30303376312E303
 30 32 44 33 30 33 37 33 34 33 30 32 44 34 36 33   02D303734302D463
 31 33 36 33 31 33 30 33 30 30 30 30 30 30 30 30   1363130300000000
 30 30 30 46 31 38 39 0D 7E 31 32 30 31 34 36 30   000F189.~1201460
 30 31 30 37 38 35 33 34 38 33 32 33 34 33 31 33   0107853483234313
 31 33 30 33 38 33 30 33 30 33 31 33 32 33 39 32   1303830303132392
 30 32 30 32 30 32 30 32 30 32 30 32 30 30 30 30   0202020202020000
 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000
 30 34 45 36 46 37 36 32 30 32 30 33 38 32 30 33   04E6F76202038203
 32 33 30 33 32 33 34 32 43 33 30 33 39 33 41 33   23032342C30393A3
 32 33 35 33 41 33 33 33 35 30 30 30 30 30 30 30   2353A33350000000
 30 30 30 30 30 30 30 30 30 30 30 30 30 45 35 46   0000000000000E5F
 35 0D 7E 31 32 30 31 34 36 30 30 37 30 39 30 30   5.~1201460070900
 30 30 31 31 30 30 43 37 39 30 43 37 44 30 43 37   001100C790C7D0C7
 38 30 43 38 30 30 43 37 42 30 43 38 32 30 43 37   80C800C7B0C820C7
 41 30 43 38 31 30 43 37 45 30 43 38 33 30 43 37   A0C810C7E0C830C7
 39 30 43 38 33 30 43 37 43 30 43 37 44 30 43 37   90C830C7C0C7D0C7
 38 30 43 37 43 30 36 30 42 36 41 30 42 36 42 30   80C7C060B6A0B6B0
 42 36 41 30 42 36 44 30 42 41 39 30 42 37 32 46   B6A0B6D0BA90B72F
 46 46 46 46 31 46 35 30 30 30 30 43 37 44 30 30   FFFF1F50000C7D00
 30 30 30 30 45 42 43 30 32 30 30 30 30 32 39 45   0000EBC02000029E
 38 30 30 30 38 30 30 30 30 32 37 31 30 32 33 44   800080000271023D
 45 32 32 0D 7E 31 32 30 31 34 36 30 30 30 30 34   E22.`

nkinnan, 2025-02-09:
Thanks for all the details!  I believe you have correctly identified these settings:

```
  protocol_commandset: 0x20   # example only
  protocol_version: 0x12    # example only
  battery_chemistry: 0x46   # example only
```

but it looks like the `protocol_variant` is wrong (or perhaps it is a variant I haven't seen before).

Did you try following this step?  Can you share the log output from it?  Does it say it has detected the variant or does it say "unable to auto-detect"?  https://github.com/nkinnan/esphome-pace-bms?tab=readme-ov-file#step-5-an-extra-step-for-commandset-20

Marcin685, 2025-02-09:
Hello,
I am attaching the esphome log from this procedure

`sensor:
  - platform: pace_bms
    pace_bms_id: pace_bms_at_address_1
    
    cell_count:
      name: "Cell Count"

text_sensor:
  - platform: pace_bms
    pace_bms_id: pace_bms_at_address_1

    system_status:
      name: "System Status"`

[logs_falownik-bateria_run_v2.txt](https://github.com/user-attachments/files/18723632/logs_falownik-bateria_run_v2.txt)

nkinnan, 2025-02-11:
If the software is talking to the powerwall over the same port that the ESP is using then those protocols shouldn't matter.

Earlier you shared the battery responses being sent back to the manufacturer software.  Can you also please share the requests that the manufacturer's software is sending out to the battery?

Marcin685, 2025-02-11:
Is this the data?

[dump_xj_bms.txt](https://github.com/user-attachments/files/18747375/dump_xj_bms.txt)

nkinnan, 2025-02-11:
Those appear to be the responses not the requests.

Marcin685, 2025-02-12:
What is an easy way to recommended requests?

nkinnan, 2025-02-14:
What operating system and what software are you using to sniff the com port?

Marcin685, 2025-02-14:
i use Device Monitoring Studio on win11

Marcin685, 2025-02-15:
Hello, I managed to read the questions and answers.

[write2a.txt](https://github.com/user-attachments/files/18811807/write2a.txt)

[read2a.txt](https://github.com/user-attachments/files/18811808/read2a.txt)