nkinnan/esphome-pace-bms #30: Not workingto address 2??

HGQ21102, 2025-02-22:
Not workingto address 2??

Setup：
//
pace_bms:
  id: pace_bms_at_address_1
  address: 2
  uart_id: uart_1
  update_interval: 5s  
  request_throttle: 100ms 
  response_timeout: 1000ms 
//

Error reporting：
[11:05:15][E][pace_bms_protocol:022]: Response from wrong Bus Id
[11:05:15][E][pace_bms:485]: Unable to decode 'read analog information' response
[11:05:15][E][pace_bms_protocol:022]: Response from wrong Bus Id
[11:05:15][E][pace_bms:501]: Unable to decode 'read status information' response
[11:05:15][E][pace_bms_protocol:022]: Response from wrong Bus Id
[11:05:15][E][pace_bms:517]: Unable to decode 'read hardware version' response
[11:05:15][E][pace_bms_protocol:022]: Response from wrong Bus Id

nkinnan, 2025-07-04:
Interesting.  Sorry for the delay in looking at this.  Would you mind please enabling verbose log mode and sending me the output?  That would help me to understand why you're seeing this error.

Haso-M, 2025-07-05:
```
RS232
1# 2# 3# 4#
address 2 ： OFF---ON---OFF---ON
```
If that's true, your address is 10 not 2.  For address 2 switch positions needs to be:
```
RS232
1# 2# 3# 4#
address 2 ： OFF---ON---OFF---OFF
```

btsimonh, 2025-07-06:
I've got my batteries linked together with RS485 (straight UTP cable).
Then, if I ask for analog from 2, but via 1, and comment out the return here:
	// Bus Id
	uint8_t addr = ReadHexEncodedByte(response, byteOffset);
	if (addr != busId)
	{
		LogError("Response from wrong Bus Id");
		//return -1;
	}

it responds with 1 as the id in the base message, but 2 in the analog info section (for a single pack request).

I can also ask for 0xFF, and get BOTH batteries.

Communications logs below.

Note: when using PaceBMS software, I can't read pack 1 from pack 2, I can only read pack 2.  It's like to do addressed request you MUST go through 1 as the rs485 master?

Single request for pack 2, send to pack 1 on 232:
```
~25024642E00202FD2E - header asks for addr 2

~25014600D07C - header says addr 1
0002 - maybe for a single request - this is pack number?
10
0D230D280D280D260D280D270D280D160D270D260D280D270D260D270D270D26
06
0BB40BB80BB40BB70BEC0BFE
07FB
D25C
218E
02
37D2
01B0
4E20
00

E281 - CS
```

single request for pack 1, send to pack 1:
```
~25014642E00201FD30

~25014600D07C - header response is addr 1
0001 - and we get 1 here
10
0D270D270D2B0D240D2A0D250D270D280D260D290D290D280D270D2A0D250D26
06
0BB70BB50BB50BB50BED0BF3
067C
D277
3AA1
02
5899
0104
5910
42

E29B - CS

```

```
multi-request via pack 1:
~25014642E002FFFD05

multi response:
~25014600D0F4 - header from pack/addr 1
0002 - 2 packs?  

// note, from the total design spec mAh, we can see this is pack 1 then pack 2.

10
0D070D080D090D050D0E0D080D100D080D030D080D060D0A0D060D060D080D02
06
0BB30BB10BB00BB10BE40BEE
01FB
D07C
382A
02 - 02 not known
5899
0104
5910
3F

// note no pack number for second item?
10
0D020D0D0D060D070D0B0D060D080CFD0D060D050D080D070D080D0A0D060D0A
06
0BB00BB40BB00BB30BE20BFA
023B
D06E
1EEB
02 - 02 not known
37D2
01B0
4E20
00

C874 - CS

```

Note: the extra byte when AnalogInformationUserDefinedValue == 2 appears to be a single byte representing SOC - but for one battery it is zero.

Haso-M, 2025-07-06:
I never checked it, but I clearly remember reading somewhere that, PACE BMS uses Modbus RTU for communication beetwen batteries over RS485 (parallel connection ports on the right side). If that's true the only way to get the state of all batteries over RS232 from single connection point is to use RS232 port on master battery. (in Modbus RTU network, master never responds to requests, only sends them).
Is it doable using PACE IC protocol over RS232, I don''t know.

RoganDawes, 2025-07-06:
I seem to recall snooping on downstream traffic, and thought it was the PaceBMS protocol that was in use. However, I would agree that it is likely only an upstream battery that can query downstream batteries. So, if you want a single device to be able to query all the attached batteries, it would logically have to be attached to the most upstream battery to do so.

As I see it, there are two possible topologies for the RS485 connections:
1. a bus. In this case, all batteries see the same traffic, and it implies that the master is actually the inverter. This also means that all batteries have to be programmed to speak the same protocol over RS485. It is then difficult-to-impossible to use RS485, RS232 and CAN (i.e. more than one) at the same time, as responses to the RS232 and CAN requests would be visible to the master on the RS485 bus.
2. a daisy chain. In this case, the most upstream battery receives queries from CAN/RS485/RS232, and then relays them to the downstream port, using whatever protocol, and then relays the answers back to the relevant interface, doing protocol translation as needed. This actually makes more sense, as it would allow there to be a RS485 "master" in relation to the most upstream battery, as well as a CAN master and an RS232 counterparty making requests at the same time, with requests being relayed hop by hop as needed to reach the necessary battery.

btsimonh, 2025-07-07:
query from pack 1, asking for data from pack 2, or asking for all packs (using 0xff) is working fine for me.  Also pbmstools is able to read both packs on RS232.  Note I'm not using this project, but instead a transpose of part of it into javascript as a node-red flow... using an RS232 USB adaptor on an RPi3.  So far I've only implemented the read analog command.

SaschaKP, 2025-09-22:
I've tried with terminal program, by sending manual commands to the chained slave battery, every command that is sent from slave for any chained battery will not be accepted/responded, only command I haven't tested is 0xFF, but as I remember, having tried it in master, the 0xFF is a special case, because the packet contains all the data of all batteries, without anything to separate them, example:
status information request
`~25FF4644E002FFFCD8`
answer
`~25014600F09800021002020202020202000202020202020202060000000000000002000000060000000005000810020202000200020202020202020202020600000000000000020000800648000000050000E087`
analog information request
`~25FF4642E002FFFCDA`
answer
`~25014600D0F40002100E1D0E1B0E1D0E180E140E190E130E080E180E190E180E200E200E160E1F0E1A060BF60BE20BF80BE80C0F0C0F0000E18D4E18024E1800C04E2064100E110E1B0E1D0E0F0E110E0E0E110E1D0E190E130E160E180E170E190E1B0E18060C090C0B0C120C1D0C020C120000E1C450D10350DC00074E2064C8A1`

since everything is pretty much glued together, you have to know the exact size you're receiving, in my case data is exactly 68 bytes (excluding starting ~ and ending \r) for analog info and 45 for status info for the single pack request.

RoganDawes, 2025-09-22:
One thing that might have to change is the structure of the component, to be more like modbus.

The problem being that there is a single UART, but multiple device addresses, and there will need to be some component arbitrating access to the UART, exactly like there is with modbus. But this will obviously be second to actually figuring out how to query the additional batteries.

nkinnan, 2025-09-22:
I believe this is the method used by either syssi here: https://github.com/syssi/esphome-seplos-bms or bmspace here: https://github.com/Tertiush/bmspace (it's been a while since I looked at them) of just sending FF and getting all battery pack's info in aggregate.  While I have taken care to validate all response lengths, outputting warnings in case of a mismatch, they have taken the approach of only sending the read analog info request (42h) and allowing the user to specify an "offset" to jump to the next set of data.

I'd prefer to be a bit more robust about it.

nkinnan, 2025-09-22:
It really just is a lot simpler to say "buy one esp per battery" :P But lets see what we can do...

SaschaKP, 2025-09-22:
anyway, data lenght is effectively two bytes, looking at the pbmstools the extrapolation is done like this:
`ushort num1 = ushort.Parse(Encoding.ASCII.GetString(this._surplusByte, 9, 4), NumberStyles.HexNumber);
if ((int) (byte) ((uint) num1 >> 12) != (int) (byte) (~((((int) num1 & 15) + (((int) num1 & 240 /*0xF0*/) >> 4) + (((int) num1 & 3840 /*0x0F00*/) >> 8)) % 16 /*0x10*/) + 1 & 15) || this._surplusByte.Length < 18 + ((int) num1 & 4095 /*0x0FFF*/) || this._surplusByte[0] != (byte) 126 || this._surplusByte[this._surplusByte.Length - 1] != (byte) 13)
  return;`

nkinnan, 2025-09-22:
You mean pbmstools right?  I think esphome-pace-bms is architected pretty reasonably :)

Don't forget that you need to calculate a CRC as well.  If that is invalid the message will be ignored.

nkinnan, 2025-09-25:
I would appreciate any/all feedback on that huge bullet list if you think I have gotten anything wrong.  But I think I have enough to start implementation.  Tomorrow.  I need to sleep :)

nkinnan, 2025-09-26:
The responding_address addition has been merged into main and the branch deleted.

nkinnan, 2025-09-26:
@SaschaKP Can you provide me with the full response frame of data from ~ to \r that you're seeing when broadcasting address FFh for CID 42 and 44? The linked code is difficult to read but this would greatly help me with implementation.  Sorry if you already shared that and I missed it, I've been looking at like 4 different issues and the threads are long at this point.  Would really appreciate it!

SaschaKP, 2025-09-26:
A plausible solution, with hardware used only for that scope, is to emulate a battery and give always the same answer, with device id increasing, up to 16, with only emulated data so to get the data we are seeking and evaluate the correct order, at least on 0x25 pace protocol, since the two batteries I have are on that one.

SaschaKP, 2025-09-26:
ALSO, would be interesting to know if a 0x20 protocol responds only in a 0x20 way, so we know that in a 0xff request we get a conforming answer in a single protocol way.

What we need, in those cases, is testers.

nkinnan, 2025-09-27:
Thanks!  That gave me a lot to chew on.  Some notes:
* You were right before, and I was mistaken about the multi-response payload bus id, there is only one at the top of the payload, and subsequent included responses do not include a bus id.
* In both your example and another that I have (for both commands 42h and 44h) the "bus id" in the payload is 2, right before the 2 responses.  This leads me to believe that in a broadcast response, that field's meaning is changed to be "response count".  Alternatively "last bus id responding" which, since the master is bus id 1, is the same thing as "response count" anyway.  This is just a theory, we need more data to confirm. 
* The payload length of your two-pack response was F4h (244).  A single response would have been 122 (actually I think that's off by 2 bytes because of the missing bus id in subsequent responses).  The maximum payload length is FFFh (4095) since it's a two-byte short and the top nibble is used for the length field self-checksum.  That means 33 of the CID 42 (analog info) requests would fit before overflow.  Which is good because some brands have a LOT of extra junk in there at the end after the standard fields :)  For CID 44 it is similar.
* The idea to emulate hardware is interesting, and I am curious what master-slave and slave-slave communications look like, but we'd have to sniff and dump those lines first to ensure we were "answering properly".  If someone wants to sniff that, I'm all in for pouring over the hex dump!  I'm virtually certain the returned data order is in pack bus id order, but always nice to double check.  I think we would need at least three battery packs in order to do that though, since the order that the master queries slaves is unknown, and how slaves pass along data is unknown.  So we'd have to sniff at least two links.
* I agree it would be really interesting to see how protocol 0x20 handles this.  The thing about 0x20 though, is that it varies massively between manufacturers (and from 0x25).  For 0x25 I am able to do nothing but check the "user defined value" at a particular offset, note the payload length for CID 42 and 44 based on that manufacturer / firmware revision as identified by the "user defined value", and ignore the "extra stuff" some manufacturers put at the end of the response.  Everything up to that extra stuff at the end is according to a single spec.  --  For 0x20 I have three different and contradictory specs and have had to split the code paths for each one, SEPLOS, PYLON, and EG4.  For 0x25 I have never needed to split the code path, or add any "variant" tag to the config yaml like I did for 0x20.  I think all the 0x25 data I've seen comes from a common firmware codebase, slightly modified by the manufacturer, but for 0x20 I believe it is (at least) three different firmware codebases, each trying to "emulate" pylon with their own "twist" on it.  It wouldn't be twice the work, it'd be four times the work if I got each of those implemented, and there may be more 0x20 variants I haven't seen yet.  Anyway, all that to say, if someone wants to dump a 0x20 multipack I'd love to look at it, but "no promises".  PRs would of course be welcomed but I'd like to focus on 0x25 for now.  The landscape for 0x20 is kind of a mess.

nkinnan, 2025-09-29:
OK, I've got slave discovery mode now...

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/nkinnan/esphome-pace-bms
      ref: "slave_discovery"
    components: [ pace_bms ]
    refresh: 1s

pace_bms:
  slave_discovery_mode: RELAY_AND_BROADCAST
```

if you want to give that a try, it should send both broadcast and relay requests in order to identify slaves, and it should also exercise some of the code changes around handling broadcast responses 

log mode very verbose, let me know what it spits out

nkinnan, 2025-09-29:
This was a pretty significant change, with some risk of regression.  Anyone who wants to test this incremental slave discovery mode change, it would be very much appreciate.  It includes almost everything needed for multi-pack except the change to init.py to handle the new yaml syntax (I am *really NOT* looking forward to that...)

nkinnan, 2025-09-29:
It may be necessary to increase the UART and pace_bms buffer size (make sure you get both) from the default 256, I'd go with 256 * pack_count to be safe, or 4096 max if you were to have 16 packs.

```yaml
uart:
  rx_buffer_size: 1024 # 256 * 4 for 4 battery packs

pace_bms:
  rx_buffer_size: 1024 # 256 * 4 for 4 battery packs
```

RoganDawes, 2025-10-20:
I have this working with some changes in progress from Nick. I also modularised the config files into `pacebms-sensors.yaml`, which is included by `pacebms-slave.yaml` and `pacebms-master.yaml` (which also contains the MASTER-specific entities). The top-level config then includes one master and however many slave configs as needed, as well as setting up things like the uart. Ideally, this would also have an include for the specific hardware device in use, and common things like `api:`, etc.

Attached, for your comments.

[pacebms.yaml](https://github.com/user-attachments/files/23000249/pacebms.yaml)
[pacebms-slave.yaml](https://github.com/user-attachments/files/23000252/pacebms-slave.yaml)
[pacebms-master.yaml](https://github.com/user-attachments/files/23000250/pacebms-master.yaml)
[pacebms-sensors.yaml](https://github.com/user-attachments/files/23000251/pacebms-sensors.yaml)

RoganDawes, 2025-10-20:
I added the deviceid: to each sensor rather, but I'm seeing strange entity naming in HA. Pack 1 ends up named something like sensor.pacebms_avg_voltage, while Pack 2 is sensor.pack_2_avg_voltage. Can't figure out why.

RoganDawes, 2025-10-21:
Also getting these:
```
[13:38:10][W][pace_bms_protocol:025]: fixme: 'Fully' protect bit? Might mean fully charged?
```
and yes, this likely does mean fully charged, as my batteries reached 100% at this point.

nkinnan, 2025-10-22:
Update pushed to the full_multi_pack branch that removes the warning log and outputs "Fully Charged" to the system status text string instead of the protection text string.  I'm glad we can confirm what that bit does, it was one of the lingering question marks due to incomplete documentation :)  

Hope to hear back on how the BMS hard resets and backing off on query speed affected your packs getting into some kind of degenerate state.  Hopefully a simple back-off via `request_throttle` will help resolve that, but if not we can investigate further.