nkinnan/esphome-pace-bms #32: Support for Greenrich U-P5000

RoganDawes, 2025-03-16:
Hi, I have tested this out against my Greenrich U-P5000 packs, and am getting some errors. I'll state up front that this is a slightly odd deployment, using the "host:" platform (running under Linux), talking to the host uart, so that *might* have some impact, but I don't think so, as *most* of the parameters are read correctly, and only a few seem to have problems.

I have used tertiush/bmspace addon in HomeAssistant up to now, and verified (using tcpdump on the TCP-serial data) that it was talking protocol 25, so left that all blank/default in the configuration file.

Here is my esphome config file, minus all of the sensors. I have simply used the sensors as documented in the README.md, so I deleted them to save space, and eliminate confusion as to whether they have been modified at all. I did see the same errors when just running with the `text_sensors:` group defined, trying to decode the status output.

```
esphome:
  name: pace-bms
  friendly_name: Pace BMS

host:

logger:
  level: VERY_VERBOSE

time:
  - platform: host
    id: sntp_time
    timezone: Africa/Johannesburg

uart:
  baud_rate: 9600
  port: "/dev/ttyUSB0"

external_components:
  - source:
      type: git
      url: https://github.com/nkinnan/esphome-pace-bms
      ref: "main"
    components: [ pace_bms ]
    refresh: 1d

pace_bms:
  id: pace_bms_at_address_1
  address: 1
  update_interval: 600s
  request_throttle: 50ms
  response_timeout: 1000ms

# standard sensors deleted to save space
```

Here is an example of the errors received:

```
[21:57:37][D][pace_bms:433]: Sending 'read analog information' request
[21:57:37][VV][pace_bms:437]: Request frame: ~25014642E00201FD30
[21:57:37][V][pace_bms:456]: Processing response frame for 'read analog information' request
[21:57:37][VV][pace_bms:460]: Response frame: ~25014600D07C0001100CCB0CCC0CCB0CCC0CCB0CCB0CCB0CCB0CCA0CCA0CCB0CCB0CCB0CCB0CCB0CCC060B9F0B9F0B9E0BA40BA10BB9FE25CCB11E280226AC02EB26AC4EE086
[21:57:37][D][pace_bms:480]: Processing 'read analog information' response
[21:57:37][W][pace_bms_protocol:025]: Response contains a constant with an unexpected value, this may be an incorrect protocol variant. This will be ignored, but please file an issue report with full logs at VERY_VERBOSE level.
[21:57:37][E][pace_bms_protocol:022]: Length mismatch reading analog information response: 2 bytes off. This will be ignored, but please file an issue report with full logs at VERY_VERBOSE level.
```

This is the complete output of esphome run host.yaml, with logging set to VERY_VERBOSE:

```
[21:57:37][I][app:029]: Running through setup()...
[21:57:37][V][app:030]: Sorting components by setup priority...
[21:57:37][VV][scheduler:063]: set_interval(name='', interval=60000, offset=4960)
[21:57:37][C][uart.host:112]: Opening UART port...
[21:57:37][VV][scheduler:063]: set_interval(name='update', interval=900000, offset=376672)
[21:57:37][VV][scheduler:063]: set_interval(name='update', interval=600000, offset=266330)
[21:57:37][I][app:062]: setup() finished successfully!
[21:57:37][VV][scheduler:225]: Running interval 'update' with interval=900000 last_execution=3244843418 (now=3246120090)
[21:57:37][VV][scheduler:225]: Running interval 'update' with interval=600000 last_execution=3245253760 (now=3246120090)
[21:57:37][V][pace_bms:105]: Queueing v25 refresh commands
[21:57:37][V][pace_bms:295]: Read commands queued: 21
[21:57:37][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=3246055129 (now=3246120090)
[... 694 lines cut ...]
[21:57:40][D][pace_bms:816]: Processing 'read mosfet over temperature configuration' response
[21:57:40][V][pace_bms.number:853]: 'mosfet_over_temperature_alarm': Publishing state due to update from the hardware: 90.000000
[21:57:40][V][pace_bms.number:858]: 'mosfet_over_temperature_protection': Publishing state due to update from the hardware: 110.000000
[21:57:40][V][pace_bms.number:863]: 'mosfet_over_temperature_protection_release': Publishing state due to update from the hardware: 85.000000
[21:57:40][D][number:012]: 'Mosfet Over Temperature Alarm': Sending state 90.000000
[21:57:40][D][number:012]: 'Mosfet Over Temperature Protection': Sending state 110.000000
[21:57:40][D][number:012]: 'Mosfet Over Temperature Protection Release': Sending state 85.000000
[21:57:40][D][pace_bms:433]: Sending 'read environment over/under temperature configuration' request
[21:57:40][VV][pace_bms:437]: Request frame: ~250146E70000FD92
[21:57:40][V][pace_bms:456]: Processing response frame for 'read environment over/under temperature configuration' request
[21:57:40][VV][pace_bms:460]: Response frame: ~25014600501A010A1409E20A140D020D340D02F823
[21:57:40][D][pace_bms:831]: Processing 'read environment over/under temperature configuration' response
[21:57:40][V][pace_bms.number:913]: 'environment_under_temperature_alarm': Publishing state due to update from the hardware: -15.000000
[21:57:40][V][pace_bms.number:918]: 'environment_under_temperature_protection': Publishing state due to update from the hardware: -20.000000
[21:57:40][V][pace_bms.number:923]: 'environment_under_temperature_protection_release': Publishing state due to update from the hardware: -15.000000
[21:57:40][V][pace_bms.number:928]: 'environment_over_temperature_alarm': Publishing state due to update from the hardware: 60.000000
[21:57:40][V][pace_bms.number:933]: 'environment_over_temperature_protection': Publishing state due to update from the hardware: 65.000000
[21:57:40][V][pace_bms.number:938]: 'environment_over_temperature_protection_release': Publishing state due to update from the hardware: 60.000000
[21:57:40][D][number:012]: 'Environment Under Temperature Alarm': Sending state -15.000000
[21:57:40][D][number:012]: 'Environment Under Temperature Protection': Sending state -20.000000
[21:57:40][D][number:012]: 'Environment Under Temperature Protection Release': Sending state -15.000000
[21:57:40][D][number:012]: 'Environment Over Temperature Alarm': Sending state 60.000000
[21:57:40][D][number:012]: 'Environment Over Temperature Protection': Sending state 65.000000
[21:57:40][D][number:012]: 'Environment Over Temperature Protection Release': Sending state 60.000000
```

nkinnan, 2025-07-04:
Thanks for the detailed report!  Looks like a slight variation on protocol 25 as you suspected.  I will make a small tweak add this as a recognized protocol variant which will eliminate the warning for you.

But before I do that, I want to make sure it's operating 100% correctly.  You said that a few parameters seemed to be read incorrectly, can you be more specific?  I have enough information from your log output to debug, I just need to know which values you believe to be wrong.  Thanks!

RoganDawes, 2025-07-05:
I was basing that purely on the errors reported in the logs, not on any data viewed in HA. My objective in using your component over tertiush/vm space was to get the switches to disable charging/discharging of the battery, in order to balance the two different batteries that I have. One discharges faster and charges slower than the other one, resulting in a discrepancy in SoC over time if both batteries are not charged to 100%.

nkinnan, 2025-09-22:
I can understand your hesitation about the MOSFETs, it took me a while to get confident enough to send those commands myself.  What I can tell you is that the commands sent "on the wire" are identical to those sent by pbmstools.  So if pbmstools works then this should as well.  If you haven't tried it in pbmstools yet, maybe try it there first, since then you can't blame me if something goes wrong! ;)  But if the one works, then so should the other.

It's not unusual for different battery packs to have slightly different internal resistances, leading to one discharging (or charging) faster or slower than another.  First check that you have good solid connection between the packs and that the terminals are screwed down tight.  If that's all good, then "it's just how it is" since the LiFePO4 battery chemistry has such a flat charge/discharge curve.  It's not necessarily a bad thing though.  Is there a reason you want to keep them in-sync manually by messing with the MOSFET control?  

Let me see if I can add proper support for your pack, thank you for sharing the very_verbose logs, it contains all the data I'll need.

nkinnan, 2025-09-22:
I have just committed a change that should add full support for Greenrich U-P5000 if you would like to try it out.  This change will also enable decoding of the status values which was previously failing.  It should be 100% working now.

If you could also confirm that you're no longer getting any warnings in the logs and that all status information is being decoded, that'd be very helpful.  I've added it as a supported model, but I just want to make sure.

nkinnan, 2025-09-22:
Ah, yes, I can understand the frustration there!  I wouldn't want to bring one of my packs that low just for longevity reasons...  Keep me updated!  I'm curious to see what solution you come up with for opening the discharge MOSFET programatically  :)

RoganDawes, 2025-09-22:
This is what I get:
```
./.esphome/build/pace-bms/.pioenvs/pace-bms/program
[10:17:32][I][app:029]: Running through setup()...
[10:17:32][C][uart.host:112]: Opening UART port...
[10:17:32][C][api:026]: Setting up Home Assistant API server...
[10:17:32][I][app:062]: setup() finished successfully!
[10:17:32][W][component:157]: Component api set Warning flag: unspecified
[10:17:32][D][pace_bms:433]: Sending 'read analog information' request
[10:17:32][I][app:100]: ESPHome version 2025.2.2 compiled on Sep 22 2025, 10:17:09
[10:17:32][C][logger:177]: Logger:
[10:17:32][C][logger:178]:   Max Level: DEBUG
[10:17:32][C][logger:179]:   Initial Level: DEBUG
[10:17:32][C][uart.host:182]: UART:
[10:17:32][C][uart.host:183]:   Port: /dev/ttyUSB1
[10:17:32][C][uart.host:191]:   Port status: opened
[10:17:32][C][uart.host:192]:   Baud Rate: 9600
[10:17:32][C][uart.host:193]:   Data Bits: 8
[10:17:32][C][uart.host:194]:   Parity: None
[10:17:32][C][uart.host:198]:   Stop Bits: 1
[10:17:32][C][pace_bms.sensor:035]: pace_bms_sensor:
[10:17:32][C][pace_bms.sensor:036]:   Cell Count 'Cell Count'
[10:17:32][C][pace_bms.sensor:036]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:036]:     Unit of Measurement: ''
[10:17:32][C][pace_bms.sensor:036]:     Accuracy Decimals: 0
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 01'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 02'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 03'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 04'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 05'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 06'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 07'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 08'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[10:17:32][C][pace_bms.sensor:038]:   Cell Voltage X of 16 'Cell Voltage 09'
[10:17:32][C][pace_bms.sensor:038]:     Device Class: 'voltage'
[10:17:32][C][pace_bms.sensor:038]:     State Class: 'measurement'
[10:17:32][C][pace_bms.sensor:038]:     Unit of Measurement: 'V'
[10:17:32][C][pace_bms.sensor:038]:     Accuracy Decimals: 3
[... 926 lines cut ...]
[10:18:03][D][number:012]: 'Discharge Under Temperature Protection': Sending state -10.000000
[10:18:03][D][number:012]: 'Discharge Under Temperature Protection Release': Sending state -5.000000
[10:18:03][D][pace_bms:433]: Sending 'read system date/time' request
[10:18:03][D][pace_bms:801]: Processing 'read system date/time' response
[10:18:03][D][datetime.datetime_entity:048]: 'System Date and Time': Sending datetime 2025-09-22 10:19:55
[10:18:03][D][pace_bms:433]: Sending 'read mosfet over temperature configuration' request
[10:18:03][D][pace_bms:816]: Processing 'read mosfet over temperature configuration' response
[10:18:03][D][number:012]: 'Mosfet Over Temperature Alarm': Sending state 90.000000
[10:18:03][D][number:012]: 'Mosfet Over Temperature Protection': Sending state 110.000000
[10:18:03][D][number:012]: 'Mosfet Over Temperature Protection Release': Sending state 85.000000
[10:18:03][D][pace_bms:433]: Sending 'read environment over/under temperature configuration' request
[10:18:03][D][pace_bms:831]: Processing 'read environment over/under temperature configuration' response
[... cut ...]
[10:18:03][D][number:012]: 'Environment Under Temperature Protection Release': Sending state -15.000000
[10:18:03][D][number:012]: 'Environment Over Temperature Alarm': Sending state 60.000000
[10:18:03][D][number:012]: 'Environment Over Temperature Protection': Sending state 65.000000
[10:18:03][D][number:012]: 'Environment Over Temperature Protection Release': Sending state 60.000000
```

Fewer errors, just this:
```
[10:18:32][E][pace_bms_protocol:022]: Error code returned by device: CID2 Undefined (unknown command)
[10:18:32][E][pace_bms:579]: Unable to decode 'read protocols' response
```

nkinnan, 2025-09-22:
The command for "read protocols" is not supported on all battery packs.  That just means your pack doesn't support changing the protocol it uses to talk to your inverter.  It will only speak Paceic v25.  You can get rid of that error by simply removing these lines from your config yaml:

```yaml
    # setting the protocol is possible on some version 25 BMSes but not all
    protocol_can:
      name: "Protocol (CAN)"
    protocol_rs485:
      name: "Protocol (RS485)"
    protocol_type:
      name: "Protocol (Type)"
```

So, all good!

RoganDawes, 2025-09-22:
All in all, it seems to be working well, although I didn't try any of the controls yet. Now I'd love to figure out how to handle chained batteries. I posted a theory on the other issue (https://github.com/nkinnan/esphome-pace-bms/issues/24#issuecomment-2727566155), wondering what the easiest way to test it is.

nkinnan, 2025-09-22:
I'm currently going through all open issues here on github and trying to get everything taken care of.  The chained battery packs question will likely be the last thing I tackle since it's a bit more involved.  I haven't forgotten though, and it is something I'd like to support.  Right now the biggest issue for me is that I don't have the ability to test.  If you'd be willing to go back and forth with me on it a bit, that would help a lot.  I'll try to get to that soon.

RoganDawes, 2025-09-22:
Absolutely! We seem to be in compatible time zones, so happy to work with you. Can even do Discord if that is easier, since I saw you there 😁