nkinnan/esphome-pace-bms #35: [Not issue] Document with more info about Pylontech protocol variant.

kitor, 2025-07-11:
Hey,
I'm implementing interface between my batteries (running very custom Pace BMS with v25 protocol with no option to chain batteries - just RS232) and chinese inverter (Vevor EML3500-24L) that supports "Pylontech" aka v20 protocol via RS485.

Your repository was a great help while implementing v25 side. I noticed in v20 code there's a lot of CID2's missing though.
Pylontech is not very well documented and I found it the hard way as my inverter was asking for 0x61 and 0x63 CIDs... but digging the web I found a very unassuming doc that explain some of the missing ones.

It is called "Sunsynk Battery Compatibility User Manual", the version I attached is v12, dated 11/11/2022. Not sure if any newer is available since their site requires you to login to access downloads.

Anyway, document lists CID2s 42,44,47,4F,51,60-64,92-96 and provides detailed description for 60-64 which seems to be used to gather "overall/average data of entire battery system".
It has some typos/mistakes, two I found:
- in 61h field 2 (total current) example shows accuracy 3 but table lists 1
- in 64h (shutdown a specific battery pack) table lists CID2 as 9Bh instead of 64h

You may find this interesting to get even more data from those batteries :)

kitor, 2025-07-11:
[Sunsynk_BatteryCompatibility_v12_English.pdf](https://github.com/user-attachments/files/21188886/Sunsynk_BatteryCompatibility_v12_English.pdf)

kitor, 2025-07-11:
Fun, just accidently found V14 dated 02/13/23, also adding it here.

[Sunsynk_BatteryCompatibility_v14_English.pdf](https://github.com/user-attachments/files/21188901/Sunsynk_BatteryCompatibility_v14_English.pdf)

nkinnan, 2025-09-22:
https://github.com/nkinnan/esphome-pace-bms/tree/main/protocol_documentation/paceic/0x20/SUNSYNK

kitor, 2025-09-22:
Hey!
For v20 - Sunsyk documents have a list of tested batteries, I would imagine those listed as RS485 have to support 61h / 63 hCIDs. Of course those are redundant for 42h / 44h so there's no point implementing those here unless there's a battery confirmed not to have 42h/44h implemented. 

Since I'm running my code for a good month now and found some edge cases, I'll just add that I found some oddities about (implemented here) things for v25 protocol too. 

1. Proper cell balancing:
I think I found this at some Youtube video - Pace will balance cells only if thresholds are met and there's a detectable charging current available. Thus balancing is far from perfect. In case of my units they don't detect charge/discharge below 0.5A and balancing stops even though by cell voltages you can see it is still taking a bit of energy in.

But there's a very simple trick to force them into full balancing - you need to trigger cell/pack overvoltage protection threshold (prot, not alarm, from WriteCellOverVoltageConfiguration D0h).
As soon as threshold is met, they will disconnect charging and go into full, proper cell balancing mode - balancing them offline all the way to params set via WriteCellBalancingConfiguration / B5h.

When this happens, battery will report Fully protection bit (fun that they implemented that as a prot) and Cell OV / Pack OV)

2.  Charge / discharge lockout after Fully is triggered and fun around Shutdown command.

This may be specific to my custom Pace BMS (Pace support confirmed they made it but will not provide any details -.-) but when Fully is hit, half of the time my batteries will not allow to discharge again until restarted. But fun part - they report discharge mosfets are still open, just charge closed.

In your code I found that WriteShutdownCommand / 9Ch reboots the pack instead of shutdown and this is indeed the case, but it seems my packs in some conditions just ignore that command. 

Later I tried to use Write**Charge**MosfetSwitchCommand 9Ah... and at least in my case this immediately cleans Fully prot (as well as Cell OT prot too!) which puts the pack back online.

nkinnan, 2025-09-22:
Very interesting info, thank you for sharing!  

I think I have seen similar behavior on my v25 battery pack to what you're describing.  From what I remember, they don't really start balancing unless I hit the warning alarm level, but I didn't need to hit the protection (cutoff the MOSFET) threshold.  Since those levels are configurable, I wonder if there's some interplay here with how things are configured.  I'm going to need to do some testing and see what I see, then update the thread.  Can you share you configuration values?  Voltage levels for warning/protection and so on?  A full dump would be great, then I can compare it to mine.

"but when Fully is hit, half of the time my batteries will not allow to discharge again until restarted. But fun part - they report discharge mosfets are still open, just charge closed." I think you may have confused open/closed - open means "disconnected" / "switch off", closed means "connected" / "switch on".  Can you confirm what you meant here?

My observation about the shutdown/reboot command is that if there is no charge or discharge current, then it will shut down, but if there is any load on the BMS, or if it is charging, then maybe it will "shut down" momentarily, but then it immediately it turns itself back on since it needs to monitor what's happening given that it's not in an idle state.  Or maybe that was my v20 pack, I'm going from memory here.  But that's why I left it named "Shutdown", since it seems that is the actual behavior, as well as the documentation saying "shutdown" - it just immediately boots again if it's not idle.

"Later I tried to use WriteChargeMosfetSwitchCommand 9Ah... and at least in my case this immediately cleans Fully prot (as well as Cell OT prot too!)" it cleared Cell Over Temperature? Or did you mean Over Voltage?  Either way that's concerning, but then not entirely unexpected to find bugs like this in Chinese firmware lol

kitor, 2025-09-22:
Sure, stock pack configuration is below (those are 24V packs). I later adjusted balance threshold do 3.32v and balance delta to 20mv, FullCharge voltage to 27.95v (so it's slightly below Pack OV alarm) and lowered those IMO very high overtemp thresholds - I never saw over 40 deg C on any sensor of my packs.

Mine spent ~2 years in storage (they were brand new, 1-3 cycles) and I had to do some stupid things to bring them back in balance.
To be exact I kept them at FullCharge voltage level and repeatedly let them hit Cell OC, balance, repeat - until I got rid of single cell going immediately higher that others. 

[image]

[image]