nkinnan/esphome-pace-bms #36: Errors reported for Eenovance and Sunsynk branded batteries sold in South Africa

johnmsole, 2025-08-12:
**Thanks for a brilliant esphome solution.**  

I have connected to the RS232 port on the Eenovance/Sunsynk 10.65kwh batteries.  I am able to retrieve most parameters but am some errors reported.  

Processing 'read analog information' response
-Response contains more temperature readings than are supported, results will be truncated
-Response contains a constant with an unexpected value, this may be an incorrect protocol variant. This will be ignored, but please file an issue report with full logs at VERY_VERBOSE level.
-Length mismatch reading analog information response: 28 bytes off. This will be ignored, but please file an issue report with full logs at VERY_VERBOSE level

Processing 'read status information' response
-Length mismatch reading status information response: 6 bytes off. This will be ignored, but please file an issue report with full logs at VERY_VERBOSE level.
-Unable to decode 'read status information' response

Full VERY_VERBOSE attached.  If needed I can take a screen grab from PMBS version provided by the vendor.

Any assistance or guidance will be appreciated.  
Thanks John

[logs_pace-1_logs (1).txt](https://github.com/user-attachments/files/21727499/logs_pace-1_logs.1.txt)

nkinnan, 2025-09-22:
I was able to get the data I needed from the UART prints (it was annoying but I was able to decode it).  There should now be support for Eenovance/Sunsynk and you shouldn't get these errors anymore.  If you could confirm that all data looks to be decoded properly with this change I would appreciate it very much.

You said:  "I have also found that when connecting directly to the second battery
(first slave) that the battery expects the address to be set as "1" but
then returns the address as "2" in the responses so I disabled that check
in your code so long and am collecting the analog data from my 3 batteries
at the moment."

This could be a firmware bug (wouldn't surprise me) or it could be more subtle.  You have all your packs daisy-chained together is that right?  But then also have an ESP on each individual pack?  I'm currently looking at adding multi-pack support but it's kind of subtle/difficult to get right.  Knowing your setup is actually an interesting datapoint for me.

You said:  "I have also tried changing the parameter in your code to allow a max of 8
temperatures readings but ran out of time trying to get that to work.
Thanks for any suggestions."

I do see that your battery pack is reporting 8 temperatures.  In the change I made to support your pack, I am skipping over the two "extra" temperature readings since I'd otherwise have to modify the "client interface" / yaml.  For a strange on-off situation like this it's probably better to just drop the two extras (I've never seen anything other than 6 on any other pack).  

Can you tell me what values you are seeing for Full Capacity, Cycle Count, and Design Capacity (and whether the numbers you're getting for those values make sense?)  This will help me understand if your pack really does have 8 temperature readouts and if I implemented support for this protocol variation correctly.

nkinnan, 2025-09-22:
Fixed a bug in the Eenovance/Sunsynk support just now, so if you pulled the update before the timestamp on this message, please try it again.

johnmsole, 2025-09-22:
Will do later tonight

nkinnan, 2025-09-22:
Thanks! If you can also give me model numbers or a link to a mfg page for your two types of battery packs, that would help me add it to the known supported models list (assuming it's working properly now!)

johnmsole, 2025-09-22:
Nick.

See log file attached,  I am still receiving errors.  I am not sure if I
need to specify a specific variance for the protocol.

J

INFO ESPHome 2025.9.1
INFO Reading configuration /config/esphome/pace-1.yaml...
INFO Updating ***@***.***
INFO Starting log output from 192.168.42.115 using esphome API
INFO Successfully resolved esp32-0x25-sensorsonly @ 192.168.42.115 in 0.000s
INFO Successfully connected to esp32-0x25-sensorsonly @ 192.168.42.115 in 0.119s
INFO Successful handshake with esp32-0x25-sensorsonly @ 192.168.42.115 in 0.121s
[19:19:10.918][I][app:185]: ESPHome version 2025.9.1 compiled on Sep 22 2025, 19:14:51
[19:19:10.921][I][app:187]: Project esphome.web version dev
[19:19:10.923][C][wifi:661]: WiFi:
[19:19:10.938][C][wifi:444]:   Local MAC: E4:65:B8:54:2A:2C
[19:19:10.940][C][wifi:449]:   SSID: 'Sole'[redacted]
[19:19:10.943][C][wifi:452]:   IP Address: 192.168.42.115
[19:19:10.964][C][wifi:456]:   BSSID: F0:9F:C2:AA:FF:14[redacted]
[19:19:10.964][C][wifi:456]:   Hostname: 'esp32-0x25-sensorsonly'
[19:19:10.964][C][wifi:456]:   Signal strength: -37 dB ▂▄▆█
[19:19:10.965][V][wifi:464]:   Priority: 0.0
[19:19:10.985][C][wifi:467]:   Channel: 11
[19:19:10.985][C][wifi:467]:   Subnet: 255.255.255.0
[19:19:10.985][C][wifi:467]:   Gateway: 192.168.42.1
[19:19:10.985][C][wifi:467]:   DNS1: 192.168.42.5
[19:19:10.985][C][wifi:467]:   DNS2: 0.0.0.0
[19:19:10.985][C][logger:273]: Logger:
[19:19:10.985][C][logger:273]:   Max Level: VERY_VERBOSE
[19:19:10.985][C][logger:273]:   Initial Level: VERY_VERBOSE
[19:19:10.993][C][logger:279]:   Log Baud Rate: 115200
[19:19:10.993][C][logger:279]:   Hardware UART: UART0
[19:19:10.995][C][logger:286]:   Task Log Buffer Size: 768
[19:19:10.999][C][uart.arduino_esp32:150]: UART Bus 1:
[19:19:11.003][C][uart.arduino_esp32:151]:   TX Pin: GPIO17
[19:19:11.006][C][uart.arduino_esp32:152]:   RX Pin: GPIO16
[19:19:11.018][C][uart.arduino_esp32:154]:   RX Buffer Size: 256
[19:19:11.022][C][uart.arduino_esp32:156]:   Baud Rate: 9600 baud
[19:19:11.022][C][uart.arduino_esp32:156]:   Data Bits: 8
[19:19:11.022][C][uart.arduino_esp32:156]:   Parity: NONE
[19:19:11.022][C][uart.arduino_esp32:156]:   Stop bits: 1
[19:19:11.043][C][pace_bms.sensor:035]: pace_bms_sensor:
[19:19:11.071][C][pace_bms.sensor:015]:   Cell Count 'Cell Count'
[19:19:11.071][C][pace_bms.sensor:015]:     State Class: 'measurement'
[19:19:11.071][C][pace_bms.sensor:015]:     Unit of Measurement: ''
[19:19:11.071][C][pace_bms.sensor:015]:     Accuracy Decimals: 0
[19:19:11.071][C][pace_bms.sensor:015]:   Cell Voltage X of 16 'Cell Voltage 01'
[19:19:11.071][C][pace_bms.sensor:015]:     State Class: 'measurement'
[19:19:11.071][C][pace_bms.sensor:015]:     Unit of Measurement: 'V'
[19:19:11.071][C][pace_bms.sensor:015]:     Accuracy Decimals: 3
[19:19:11.091][C][pace_bms.sensor:025]:     Device Class: 'voltage'
[19:19:11.091][C][pace_bms.sensor:015]:   Cell Voltage X of 16 'Cell Voltage 02'
[19:19:11.091][C][pace_bms.sensor:015]:     State Class: 'measurement'
[19:19:11.091][C][pace_bms.sensor:015]:     Unit of Measurement: 'V'
[19:19:11.091][C][pace_bms.sensor:015]:     Accuracy Decimals: 3
[19:19:11.100][C][pace_bms.sensor:025]:     Device Class: 'voltage'
[19:19:11.114][C][pace_bms.sensor:015]:   Cell Voltage X of 16 'Cell Voltage 03'
[19:19:11.114][C][pace_bms.sensor:015]:     State Class: 'measurement'
[19:19:11.114][C][pace_bms.sensor:015]:     Unit of Measurement: 'V'
[19:19:11.114][C][pace_bms.sensor:015]:     Accuracy Decimals: 3
[19:19:11.115][C][pace_bms.sensor:025]:     Device Class: 'voltage'
[19:19:11.125][C][pace_bms.sensor:015]:   Cell Voltage X of 16 'Cell Voltage 04'
[19:19:11.125][C][pace_bms.sensor:015]:     State Class: 'measurement'
[19:19:11.125][C][pace_bms.sensor:015]:     Unit of Measurement: 'V'
[19:19:11.125][C][pace_bms.sensor:015]:     Accuracy Decimals: 3
[19:19:11.138][C][pace_bms.sensor:025]:     Device Class: 'voltage'
[19:19:11.152][C][pace_bms.sensor:015]:   Cell Voltage X of 16 'Cell Voltage 05'
[19:19:11.152][C][pace_bms.sensor:015]:     State Class: 'measurement'
[... 744 lines cut ...]

[19:19:39.900][D][pace_bms:801]: Processing 'read system date/time' response
[19:19:39.913][V][pace_bms.datetime:021]: 'bms_date_and_time': Publishing state due to update from the hardware: 2025:09:22 19:23:40
[19:19:39.935][VV][api.service:012]: send_message text_sensor_state_response: TextSensorStateResponse {
[19:19:39.935][VV][api.service:012]:   key: 412222983
[19:19:39.935][VV][api.service:012]:   state: '2108434A1302209Z    SQLR051208103C409329'
[19:19:39.935][VV][api.service:012]:   missing_state: NO
[19:19:39.935][VV][api.service:012]:  }
[19:19:39.945][VV][api.service:012]: send_message text_sensor_state_response: TextSensorStateResponse {
[19:19:39.945][VV][api.service:012]:   key: 412222983
[19:19:39.945][VV][api.service:012]:   state: '2108434A1302209Z    SQLR051208103C409329'
[19:19:39.945][VV][api.service:012]:   missing_state: NO
[19:19:39.945][VV][api.service:012]:  }
[19:19:39.964][D][datetime.datetime_entity:048]: 'System Date and Time': Sending datetime 2025-09-22 19:23:40
[19:19:39.985][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:30:30:34:30:30:43:31:39:30:39:31:36:31:33:31:37:32:38:46:42:32:37:0D
[19:19:40.071][VV][api.service:012]: send_message date_time_state_response: DateTimeStateResponse {
[19:19:40.071][VV][api.service:012]:   key: 276240505
[19:19:40.071][VV][api.service:012]:   missing_state: NO
[19:19:40.071][VV][api.service:012]:   epoch_seconds: 1758569020
[19:19:40.071][VV][api.service:012]:  }
[19:19:40.082][VV][api.service:012]: send_message date_time_state_response: DateTimeStateResponse {
[19:19:40.082][VV][api.service:012]:   key: 276240505
[19:19:40.082][VV][api.service:012]:   missing_state: NO
[19:19:40.082][VV][api.service:012]:   epoch_seconds: 1758569020
[19:19:40.082][VV][api.service:012]:  }

nkinnan, 2025-09-22:
OK, so this is interesting.  It "appears as if" my change didn't take, but I'm also starting to wonder if those serial prints are accurate.  When I reformat and hex decode them, I see that the read analog information response is 12 bytes longer than expected (actually 4 bytes accounting for the two extra temperatures) but the log line `Length mismatch reading analog information response: 28 bytes off.` seems to indicate that I am not seeing the full thing.  

I wonder if the uart debug print is length limited or something.  I also wonder if the reason my normal prints aren't showing up is because of the excessive length, or possibly because the logs buffer is too full to be able to accept the entire thing since such an excessive amount of logs are being generated.

As a debug step, would you please make the following changes to your YAML and send me the VERY_VERBOSE log output again?
1) Remove all sensors except for:
```yaml
sensor:
  - platform: pace_bms
    pace_bms_id: pace_bms_at_address_1

    full_capacity:
      name: "Full Capacity"
    design_capacity: # not available on EG4 protocol 0x20 variant
      name: "Design Capacity"

text_sensor:
  - platform: pace_bms
    pace_bms_id: pace_bms_at_address_1

    system_status:
      name: "System Status"
    configuration_status:
      name: "Configuration Status"
```
2) Ensure that refresh is set to 1s here:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/nkinnan/esphome-pace-bms
      ref: "main"
    components: [ pace_bms ]
    refresh: 1s
```
3) Turn off the uart debug output (just to get cleaner logs, we don't want to dump a lot of data over the API connection since I suspect this is part of the reason those log lines I really  need are missing)
4) Increase uart rx buffer size:
```yaml
uart:
  rx_buffer_size: 1024
```

nkinnan, 2025-09-22:
I've confirmed with the devs that the max log line length is 512 bytes, and that uart print is 450 due to the terrible way it's printed double-encoded with colons, and there's easily 50 or 60 bytes overhead.  So I'm seeing a truncated value for the response.

We really need to get those missing log lines to show up since it's impossible for me to fully debug and implement this without being able to see the full value that's being received from the BMS.  I'm hopeful that the three changes I asked you to make will help us to accomplish this.

nkinnan, 2025-09-25:
I suppose I can add a config option to ignore that error.  I can't just make it always ignored though.  Imagine you have a situation where battery packs are chained together and the "wrong one answers".  You'd want to know that...  It is probably a firmware bug though, as you suspect.

I'd love if you could help me understand the behavior a little bit better.  You said you have three battery packs.  What are the address dip switch settings?  Could you send me a picture of them?

Remember, I have not gotten a single picture from you (you have sent several) because GitHub *WILL NOT* save pictures sent by email.  You will have to upload the picture through github.com directly, using a web browser.  https://github.com/nkinnan/esphome-pace-bms/issues/36  This is the ONLY WAY I can view any pictures that you share.

If you wouldn't mind:
1) Can you share pictures of the dip switches?
    * Picture of pack 1 dip switches
    * Picture of pack 2 dip switches
    * Picture of pack 3 dip switches
2) What responds to what?
    * While chained together, pack 1 responds to a request for address X with address Y
    * While chained together, pack 2 responds to a request for address X with address Y
    * While chained together, pack 3 responds to a request for address X with address Y
3) How does this change if you un-chain the packs?  (remove all the "ethernet cables" between them)  Please note that you will need to reset the packs after you do this, either by opening and closing a breaker, or hitting a recessed "reset" button with a push-pin or something.
    * With packs separated (and AFTER RESETTING them manually) pack 1 responds to a request for address X with address Y
    * With packs separated (and AFTER RESETTING them manually) pack 2 responds to a request for address X with address Y
    * With packs separated (and AFTER RESETTING them manually) pack 3 responds to a request for address X with address Y

This would help me understand better what's going on.  If you could re-sent the picture of the nameplate that'd be great too!

If we can confirm this really is a firmware issue we can't work around, I'll go ahead and add support for ignoring the incorrect responding address.

nkinnan, 2025-09-25:
closing as complete but we can still chat about the address issue here

closed 2025-09-25 by nkinnan

johnmsole, 2025-09-25:
1. Can you share pictures of the dip switches?
Picture of pack 1 dip switches
Picture of pack 2 dip switches
Picture of pack 3 dip switches

Nick these batteries do not have any dip switches.  They establish their address's automatically on powerup.  The first one becomes the master which in my case is the newest battery (Eenovance).

I'll share the rest of the info requested asap

nkinnan, 2025-09-25:
Oh! That could help explain the problem :)

nkinnan, 2025-09-25:
Couldn't sleep, added the feature you need, but haven't tested it yet.  You can give it a try though.  I updated the documentation: https://github.com/nkinnan/esphome-pace-bms/blob/main/README.md#uart-and-pace_bms (look for "responding_address").

To get this change, you'll have to modify your external_components like this:

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/nkinnan/esphome-pace-bms
      ref: "add_responding_address"
    components: [ pace_bms ]
    refresh: 1s
```

Note the `ref: "add_responding_address"`, I put this in a branch since it's a pretty large change and it's untested.

johnmsole, 2025-09-25:
Great thank Nick  Ill test it now and revert

J

nkinnan, 2025-09-25:
Please share the [pace_bms] node from your YAML and let me know which battery you were querying (is the esp connected to battery 1, 2, or 3?)

nkinnan, 2025-11-09:
Thanks for the logs, I believe there was a small bug introduced in my change to enable multi-pack which caused your Eenovance (rebadged Sunsynk) not to be detected properly.  That should now be fixed.  Can you confirm that you no longer get "Response contains a constant with an unexpected value" after rebuilding and you're good to go in single-pack mode again?

If that's solved, then it may have also fixed the problem you were having with multi-pack, so please go ahead and give that a try again as well (if it's working again in single-pack mode).  Let me know the results!

nkinnan, 2025-11-09:
It's those extra temperature readings again, I think I've fixed it for the analog responses based on your logs.  There may still be a problem with the status response multi-pack decoding though, so please give it a try and share any additional error logs if you get them.

nkinnan, 2025-11-09:
I just added full support for the extra 2 temperature readouts so you can expose them in your yaml config.  I'll wait to commit that change until you let me know that multi-pack is working for you now (I don't want to have multiple changes being tested at the same time).

nkinnan, 2025-11-09:
Excellent!  Since that's working, I've also just committed the change to enable you to read those extra two temperature sensors that your variant has:

```yaml
sensor:
  - platform: pace_bms
    temperature_07:
      name: "Eenovance extra temperature 1"
    temperature_08:
      name: "Eenovance extra temperature 2"
```

I have no idea what those extra two sensors are sensing, but if you end up opening your pack to find out, let me know :)

I think that's the last thing, you should be 100% up and running.  As always, please let me know if you hit any issues with this final change.  If not, I'll consider this complete.

nkinnan, 2025-11-09:
It means a lot to me to have my work being used and appreciated.  Thank you :)